offspot/base-image #12: Crash of builder just afteer extracting packages

kelson42, 2022-06-02:
```
I: Extracting mount...
I: Extracting ncurses-base...
I: Extracting ncurses-bin...
I: Extracting passwd...
I: Extracting perl-base...
I: Extracting sed...
I: Extracting sysvinit-utils...
I: Extracting tar...
I: Extracting tzdata...
I: Extracting util-linux...
I: Extracting zlib1g...
W: Failure trying to run: chroot "/pi-gen/work/offspot-base/stage0/rootfs" /bin/true
W: See /pi-gen/work/offspot-base/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/pi-gen/work/offspot-base/stage0/rootfs/debootstrap': Directory not empty
[14:02:41] bootstrap failed: please check /pi-gen/work/offspot-base/stage0/debootstrap.log

real	6m1.728s
user	0m0.129s
sys	0m0.146s
DEBUG:builder:Removing existing container pigen_work
pigen_work
```

rgaudin, 2022-06-02:
This happens during the pi-gen build which we haven't modified at this stage.

I'd suggest you try to run vanilla pi-gen build following:

```
git clone --depth 1 --branch 2022-04-04-raspios-bullseye https://github.com/RPi-Distro/pi-gen.git pigen-repo
cd pigen-repo
./build-docker.sh
```

If that doesn't work, it might be interesting to try outside docker (`./build.sh`) but that would obviously require some dependencies that you may not want to install and you should follow [their doc](https://github.com/RPi-Distro/pi-gen/)

kelson42, 2022-06-02:
Upstream ticket open by @fheslouin at https://github.com/RPi-Distro/pi-gen/issues/590, but solution still not clear.

kelson42, 2022-06-02:
Solution is described here https://github.com/RPi-Distro/pi-gen/blob/master/README.md#binfmt_misc

rgaudin, 2022-09-21:
I believe this is addressed by the README.

closed 2022-09-21 by rgaudin