offspot/base-image #3: Add data partition

rgaudin, 2022-06-02:
We want to use a three partitions scheme with:
- an ext4 “rootfs” (already in raspian) partition for the system
- rootfs partition gets an extra 300Mi? from base-image disk usage
- an exfat “data” partition for image, containers and content storage.
- an in-file ext4 filesystem on `/data/docker.fs` to host the containers storage file.

exFAT allows anyone to plus the card on a computer and mount the data partition (Linux, macOS, Windows).
This allows for content updates in some cases and eases maintenance.

It also allows the image-creator to be multi-platform as its role consists of mounting data partition
to download image tarballs and content inside.

exFAT comes with a few limitations though:
- no permissions support. filesystem is mounted with a set user, group and mask which is applied to all the files. This prevents permissions-sensitive tools from working off of it such as the containers task. That's the reason we are storing the containers files within an in-file fs itself in ext4.
- no resize support (only on Windows to be more precise). This means we need to know the actual target SD card size when building the image. image-creator will thus require the target SD card size as input as it will resize the partition (not the FS) and recreate the filesystem before installing the content and images on it.

fheslouin, 2022-08-02:
Sadly, https://github.com/offspot/base-image/issues/4 and https://github.com/offspot/base-image/issues/5 did not success. I guess ownership and permissions management will happen with other WebApp as most of them needs to control these parameters. 

My first thought is to switch to a regular `ext4` partition, but it is not going to work for your scenario. We could create an other `ext4` volume along side the `docker.fs` volume but space management is going to be a nightmare...

rgaudin, 2022-08-02:
Since we'll only want extfs filesystems and that those FS support growing without even being unmounted ; we can imagine growing loop-device FS dynamically.
We could also imagine having per-app loop devices if that provides additional benefits (?)

It's not clear in your tickets about which files needs specific permissions. My question being: do we only need app-files (source-code) or content files (those that will be downloaded) to be on permission-aware fs.

Two possibilities:
- image-creator to mount the new _growable_ FS on the host and copy download content files inside it directly. Linux-only, so reserved for OLIP scenario.
- startup script that moves content from exfat partition into the new loop-device mountpoint. That imposes some space restrictions on apps that needs this obviously.

fheslouin, 2022-08-02:
The dynamic growing FS is an option eventough space management can be tricky on a daily basis.
I'm wondering how OLIP will react with a typical Wikipedia content installation. Usually OLIP looks for space available on `/data/ ` shared folder before doing anything. In this case space might be less than needed. 

> My question being: do we only need app-files (source-code) or content files (those that will be downloaded) to be on permission-aware fs.

Mainly app-files

commit 41b5800d (2022-07-08): Fixed #3: Add exfat data partition

export-image's prerun.sh has been rewritten to add an additional partition that is formatted to exfat.
It is mounted at `/data` and contains a `master_fs` flag file.

rootfs margin is set to 384Mi so it has a small space to use should it be needed ; since it won't be resized anymore.

A new `docker.fs` file is created on the third partition. It holds an empty ext4 filesytem that we'll be able to use for docker storage. Its size is configurable and default to 100MB. That docker fs is mounted at `/data/docker`

During build, exfat support is provided by exfat- [... cut]
commit 492e165e (2022-07-08): Fixed #3: Add exfat data partition

export-image's prerun.sh has been rewritten to add an additional partition that is formatted to exfat.
It is mounted at `/data` and contains a `master_fs` flag file.

rootfs margin is set to 384Mi so it has a small space to use should it be needed ; since it won't be resized anymore.

A new `docker.fs` file is created on the third partition. It holds an empty ext4 filesytem that we'll be able to use for docker storage. Its size is configurable and default to 100MB. That docker fs is mounted at `/data/docker`

During build, exfat support is provided by exfat- [... cut]
commit 627f50b9 (2022-07-25): Fixed #3: Add exfat data partition

export-image's prerun.sh has been rewritten to add an additional partition that is formatted to exfat.
It is mounted at `/data` and contains a `master_fs` flag file.

rootfs margin is set to 384Mi so it has a small space to use should it be needed ; since it won't be resized anymore.

A new `docker.fs` file is created on the third partition. It holds an empty ext4 filesytem that we'll be able to use for docker storage. Its size is configurable and default to 100MB. That docker fs is mounted at `/data/docker`

During build, exfat support is provided by exfat- [... cut]