offspot/base-image #20: Reduce writes to disk

rgaudin, 2022-06-04:
In order to preserve flash card storage and keep the root partition size under control, we want to reduce the number of writes to the minimum.

This implies disabling all logs that we don't need and setting strict logrotate configurations for those we want to keep – should there be any.

This doesn't concern docker applications which are handled both at application's code level and within #6.

The option we discussed together is the one followed by @fheslouin in caff30f29580b1ef86dbc7d47b5344e4e974f336 which consist of using tmpfs for /tmp and most of /var. Giving it a second though, it seems fragile as:
- it consumes RAM which is a limited resource ; especially on low-end devices like Pi0 (512M)
- limiting this effect would mean limiting writes which goes in the same direction. Better have low logs writes on SD and not using RAM.
- important logs are generated in containers and stored on content partition (http server) anyway.

@fheslouin @letompouce, what do you think?

fheslouin, 2022-06-08:
I agree that it could to be an issue on Pi0, I wonder if their is a trick to flush logs if it goes above a threshold ?! This could helps 
Probably this feature should be based on field experience and thus enable if we observe SD Card failure.

To sum up, I would say that we could reduce at most log write and keep a close look on this.

What do you think @letompouce ?

rgaudin, 2022-07-22:
[journald](https://wiki.archlinux.org/title/Systemd/Journal) must also be configured independently.

PR #40 (2022-10-05): Reduce disk writes
Fixes #20

Reducing disk writes mostly through logs management:

## disabling docker's own log store

We're instructing docker to send logs to journald so it's not storing files for it but `docker logs` can still be used.
Logs are also visible using `journalctl`

## disabling syslog

most of the logs should be going through systemd but regular syslog can still happen.
We are disabling the storage of syslog logs (so it's not affecting apps) by sending
all of them to the void.

## contain journald

main log store being systemd's journald, we are instructing it to use volatile  [... cut]
closed 2022-10-05 by rgaudin