offspot/base-image #21: Add minimum maintenance tools

rgaudin, 2022-06-04:
For development, debug and maintenance, it is occasionally required to log-into the Pi. To ease with the common tasks associated with such operations, the following packages, installed on the host, would help

- `vim-tiny` ~2.5M (`micro` is way larger)
- `screen` ~1M

FYI, `bash-completion`, `git`, `curl`, `wget`, `unzip`, `zip`, `p7zip-full` are already part of the distrib.

rgaudin, 2022-06-08:
Agrees ; those are all useful for maintenance/debug. `tmux` is indeed a better choice than `screen`. I'd suggest `ncdu` as well.

We should maybe disable updatedb's cron entry as maintainer could run it when needed and it would satisfy #20

fheslouin, 2022-06-09:
+1 for `ncdu` and disabling updatedb's cron entry

letompouce, 2022-09-14:
Oops I'm late on this one!

These two packages sound like a must-have to me:
* `ncurses-term` because not everyone has `TERM=xterm` :-) (4,2Mb)
* `jq` to query/filter any JSON output/files (1,1Mb)

This one is just a shell convenience but is so lightweight:
* `tree` for filesystem exploration (118Kb)

rgaudin, 2022-09-16:
@letompouce added them three 👍

PR #35 (2022-09-16): some host-maint tools
A few useful (and lightweight!) tools for maintenance. `screen` replaced by `tmux`.
`vim` removed as `vim.tiny` is already included (as well as `htop` and `ncdu`

Fixes #21
commit 91e469a9 (2022-09-16): Fixed #21: adding a few host-maint tools

A few useful (and lightweight!) tools for maintenance.
`screen` replaced by `tmux`.
`vim` removed as `vim.tiny` is already included (as well as `htop` and `ncdu`
closed 2022-09-16 by rgaudin