offspot/base-image #29: System Configurator

rgaudin, 2022-07-12:
System configurator is the script(s) that configures a running system upon startup. It is responsible for many features and was initially described in https://github.com/offspot/image-creator#json-configurator

It is not 100% clear where this tool should be developed, stored in the image and at which stage (base-image, image-creator) it should be added/updated if not both.

For the moment, let's have it in-here, store it in `/usr/local/bin` and add it to the base-image. Time and experience will tell if this is appropriate or not.

Initial features:

- [ ] hostname
- [ ] timezone
- [ ] network
- [ ] wireless AP
- [ ] dns
- [ ] dhcp server

kelson42, 2022-07-13:
@letompouce @fheslouin Could you please give a feedback on this? This would be something to do next.

rgaudin, 2022-10-05:
Closed in #36

closed 2022-10-05 by rgaudin