offspot/base-image #32: Docker purge strategy

rgaudin, 2022-07-26:
With #31, we now have a docker system in place but there's no purge strategy.

What a common purge strategy should be, if any, is not obvious. Given we are not connected, we won't be able to pull so evicting images should not be taken lightly.

For Kiwix scenario, I believe we are fine with zero purging. We'll only add the images we need at creation time and docker-compose will create/reuse the corresponding containers.

OLIP orchestrate images and containers so the scenario is probably more complex but in this case I believe OLIP does (or would do) the cleanup.

In the end, I am suggesting to **not add any automatic purging whatsoever** but to agree/document that it's on purpose.

kelson42, 2022-07-26:
@rgaudin No way with current setup to get deprecated/zombie containers?

rgaudin, 2022-07-26:
Not if we only create containers once via image-creator and the containers are named. Images could be loaded and not used but that's also under our creator responsibility.

kelson42, 2022-07-27:
I think @rgaudin has pretty much answered the question and I'm in favour to do nothing regardind image/container purging as long as no problem or other findings pop up.

fheslouin, 2022-07-27:
OLIP does not purge any webapp image once they have been stopped, so we can end up with several old one. 

We can either purge from host device with a purge strategy or we can try to trigger this purge from the BSF Agent container through the Docker socket or even better implement such strategy directly in OLIP.

letompouce, 2022-08-02:
The planned strategy in OLIP is [to remove dangling images](https://gitlab.com/bibliosansfrontieres/olip/olip-deploy/-/issues/20).

rgaudin, 2022-09-14:
Closing as we've agreed that there's nothing to do at base-image level at the moment.

closed 2022-09-14 by rgaudin