offspot/base-image #34: OLIP WebApp & ExFAT issue

fheslouin, 2022-08-03:
We have to kind of WebApp : 

 * The one that needs to manage ownership and permissions on folder mount from host device (MariaDB, NextCloud)
 * The one that needs to create symbolic link from container to mounted folder from host device (OLIP Mediacenter)

Currently OLIP use `/data/` on host device and share it with all WebApp and each WebApp has is own folder within `/data/`.

As discussed earlier, a possibility is to create an other ext4 loop device to host `/data/`, we still have to discuss the way to implement it.

This issue is a follow up of #4, #5 and #28

letompouce, 2022-08-03:
OLIP is mostly running WebApps. The whole webdev context is quite picky about ownership, file modes and the like ; yet there is no enforcement about such a rule and any app is free to check for those - or not. Our tests with Moodle and NextCloud confirm this.

For now, OLIP stores the WebApp datas in their own directory at `/data/webapp.app/`. This is problematic since `/data` is ExFAT, mounted with fixed ownership and modes.

Creating more loopdevices for every app tends to bring a huge layer of complexity.

Instead, a possible solution would be that the image-creator stage had a switch when it comes to recreate/extend the `/data` partition:
* for Kiwix scenario:
  * the `docker.fs` loopdevice is extended to a "fixed" size
  * some resizing system allows to resize it should it host more and more apps
* for OLIP scenario:
  * the `docker.fs` loopdevice is extended to the maximum available size
  * the resizing system would then be used once and never have anything to process anymore (it could even be deactivated by the bsf-agent)
  * OLIP would then store the WebApp data inside this `docker.fs` volume, i.e. `data/docker/olip/webapp.app`. This path is to be overriden in the docker-compose file, which is meant to be tweaked anyway.

This way, OLIP devices would have a filesystem that fulfills our needs ; the Kiwix scenario doesn't require any change.

The only downsides we can think of are:
* OLIP scenario would have a meaningless path (`/data/docker/olip/`) ; this is only cosmectic, and could be worked around by bind-mounting it to `/data/olip` for the sole sake to make our tab-completion muscle memory more instinctive
* OLIP scenario has ext4 over ExFAT, which is useless ; on the other side, it doesn't hurt and has a very very low footprint

kelson42, 2022-09-30:
What is thw status here? Accepeted thst OLIP will benefit of a full ext4 fs? Does that need a PR?

rgaudin, 2022-09-30:
🤔 ⁇ We've discussed this issue several times over the last week. I'll lay it our for history's sake.

- work has begun on this issue. 
- expected outcome is a mechanism to have different sized virtual fs depending on the scenario (olip/kiwix)
- implementation lead to creation/resizing in the _before-first-boot_ script as we need extfs support to create the extfs virtual fs.
- creating/expanding the virtual fs file on an exfat one unveiled an issue. As [exfat doesn't support sparse file](https://learn.microsoft.com/en-us/windows/win32/fileio/filesystem-functionality-comparison) the file has to be written entirely which would mean **hours of writing an empty file** in the olip scenario. That's not acceptable of course.
- circumventing this lead to supporting the `/data` partition in extfs at least for OLIP (with a virtual extfs on top or not was not settled)
- Kiwix scenario would still be **minutes** (depending on HW) during a feedbackless step.
- this lead to a discussion on kiwix-side to drop exfat support and assume linux-only for the image-creator.
- tests have been done to evaluate whether/how a linux-only image-creator could still be used in a multi-platform kiwix-installer

This is stalled until we reach a decision (we are scheduled to discuss this today!) which would either resume work on this ticket or close it and revert exfat changes.

rgaudin, 2022-10-01:
We've eventually decided to switch to `extfs` and thus drop `exFAT` support. `/data` partition will be `ext4` and we'll remove the virtual fs.

closed 2022-10-01 by rgaudin