offspot/base-image #54: Reduce memory footprint

rgaudin, 2023-09-12:
Based on #45, we target the following devices:

- Pi 3A+ (512MB)
- Pi 3B (1GB)
- Pi 3B+ (1GB)
- Pi 4B (1GB, 2GB, 4GB, 8GB)
- Pi 400 (4GB)
- Pi Zero 2W (512MB)

Of this list, 2 devices have 512MB RAM: Pi 3A+ (arguably old) and Zero 2W which is a recent device that respond to specific use cases.

An off the shelf base-image currently leaves only 128MB free RAM for all applications. That's not enough to support our baseline of tools (captive-portal, home, metrics) and leave enough for content apps. Reminder: we're using Docker so a lot of stuff are duplicated.

A Quick look at `htop` shows a great deal of RAM is used by `thttpd` so maybe that's the main culprit here; although that's strange since its job is so limited. Given it's only used in the base-image to serve as a demo, we can't really count it as base-image usage.

More investigation is needed with actual data.

kelson42, 2023-09-12:
@rgaudin I'm worried to read that metrics would be a mandatory package. I think it should not.

rgaudin, 2023-09-12:
> @rgaudin I'm worried to read that metrics would be a mandatory package. I think it should not.

You misread. I mentioned “baseline of tools” listing it in opposition to content. Neither those kiwix tools nor content are of concerned in this repo. I just want to emphasis that the _bare system_ is already using a lot of RAM

rgaudin, 2023-10-27:
Now that we're based off bookworm, podman is available in a recent-enough (4.3.1) package. We may look into #7 again if docker turns out to be an issue on Pi Zero2W

kelson42, 2023-10-27:
@rgaudin Being impatient. Anything stopping us to make right now a wikimed image and test on this HW?