offspot/base-image #62: Change WiFi firmware to support more clients

rgaudin, 2024-02-06:
Stock WiFi firmwares support too few number of clients for use in our Hotspot.
We need to use specific firmware versions with optimized number of clients.
Those are available in IIAB's repo

kelson42, 2024-02-06:
Seems very important indeed

rgaudin, 2024-02-11:
Fixed at:

- `base-image` (`1.2.0`)
  - copying all known firmwares to `/lib/firmwares/cypress`
  - backuping original firmware to `xxx_raspios`
  - symlinking firmwares to the best known ones
- `offspot_runtime` (`1.10.0`)
  - new `firmware:` option on [YAML config](https://github.com/offspot/offspot-config/tree/v1.10.0?tab=readme-ov-file#offspotyaml-format)
  - on start, if firmware change is requested, symlinks are updated
  - if firmware changed, reboot is triggered immediately

- This is very similar (but simpler) to what IIAB does.
- We reused IIAB's hand-picked firmwares

rgaudin, 2024-02-11:
We default to the 30-supporting for Pi3/ZeroW and 24 for the 3B+/4/5. I tested with 7 different devices manually (4 laptops, 3 phones) which is already above the _stock_ (raspios) limit (4/5).

closed 2024-02-11 by rgaudin