offspot/base-image #68: Make hotspot directly pluggable onto screen

rgaudin, 2024-06-04:
In GitLab by @Popolechien on Nov 1, 2019, 10:31

So we have a use case where everyone in the room has a tablet or smartphone but the teacher also wants to showcase on a large screen (projector or TV screen). Ideally, we could/should be able to simply plug the hotspot onto it and have visual display.

benoit74, 2024-10-24:
I wanted to experiment with this, and I finally more or less achieved to make it work.

Sample code for docker-compose yml (`/etc/docker/compose.yml`, config below is for demo image or any hotspot named `wikipedia`):
```
services:
  browser:
    image: bh.cr/balenalabs/browser-aarch64:v2.9.0
    privileged: true # required for UDEV to find plugged in peripherals such as a USB mouse
    environment:
      SHOW_CURSOR: 1
      LAUNCH_URL: http://wikipedia.hotspot
```

I also had to edit the /etc/hosts inside this container to add proper DNS records (DNS to the hotspot is not working from inside the container) ; I just added wikipedia.hotspot and kiwix.wikipedia.hotspot pointing to 192.168.2.1 for my tests, more entries are obviously needed, and this does not survive a hotspot restart.

The image size is not negligible (I don't have precise numbers, dive is not working for some reason on my machine with this image ... but it is more than 600M).

The whole stack is pretty slow on a Raspberry Pi 3 Model B, but still usable if you are a bit patient. Probably drains quite a lot of battery. I can't check how it runs on a Pi 5 ... because I don't have proper HDMI cable ... stay tuned!

rgaudin, 2024-10-25:
Great ; thank you for the tests!

> I also had to edit the /etc/hosts inside this container to add proper DNS records

Indeed, the per-app subdomains are unknown to docker and thus not managed by it. We can add `dns` option for this if it is required for this one.

benoit74, 2024-10-25:
Didn't knew about this `dns` option of Docker compose ... would have been way simpler indeed ^^

benoit74, 2024-10-27:
Some more experiments:
- adding `dns` to the container configuration does not help
- setting `network_mode: host` to the container configuration solves the networking issue (after all, it is already priviledged, so sharing network with host is not that a big deal
- for this to work on a Pi5, one has to set environment variable `BALENA_DEVICE_TYPE: raspberrypi5` (see `start.sh` modification done at https://github.com/balena-io-experimental/browser/pull/166/files)

I'm pleased to announce this works like a charm on a Pi5, very fast even with just an SD card. You just need a real power source otherwise your Pi will reboot / shutdown automatically when the "browser" container starts.

kelson42, 2024-10-27:
We should make a PR IMO, and see how to achieve to display the browser fullscreen (which issue is that?) to make a kiosk.

sj80231, 2024-11-06:
> We have talked so far about technical challenges, but actually the requirements of a working solution have not been clearly written. Here they are from my PoV:
>
> Once the HDMI screen plugged and the web services started, then the Kiwix Hotspot dashboard should be displayed (and all Kiwix Hotspot made available like via wifi)
> If no HDMI device is visible, then the kiosk service should be stoped (it is important that no memory is used for this if not used)
> This is a standart service of Kiwix Hotspot (not an optional one)

Hello guys,  I have been looking forward to using a fully-customizable Kiwix Hotspot for some time now, and I have seen a number of updates from GitHub, but what is the current status?  The instructions on the main website had caused some confusion for me earlier this year, and I invested a bunch of time trying to get the hotspot feature to function as described, but I wasn’t ever able to get it to work.If the feature is active now, and you have a list of steps I can follow to get the hotspot running I’d really appreciate it if you would share those with me so I can get the hotspot set-up properly and start using it.Thanks to everyone for expending the time and energy to make the hotspot feature available.  Having a fully-customizable hotspot for Kiwix will be tremendous resource.In advance, I appreciate help on this.SJ80231

rgaudin, 2024-11-07:
@sj80231 this is not yet available. Use the subscribe button on the right pane of this ticket to be notified of updates. Once this ticket is closed, it means it is implemented.

Popolechien, 2024-11-07:
@sj80231 The status is exactly what is here: we have an idea of how we get to what we want, and know what the final outcome should be. Next thing you should look for is for this issue to be closed, which will mean the feature has been pushed to the production branch. Right now no code to implement the feature has been written.

I have one question though for you:

The website states explicitely:
> Q: Can I plug my Raspberry to a screen?
> A: No, it’s not a desktop computer but a wi-fi hotspot. If you were to plug it to a screen you would probably see a command line screen (and no content).
> The only way at this stage to access your content is to connect to the wi-fi network created by the Raspberry.

So what do you mean when you write "The instructions on the main website had caused some confusion "

sj80231, 2024-11-08:
In answer to your question.  I had problems with verbiage on pages that made me believe the hotspot was a functional thing.From what I see at this point, I think you took my advice and cleaned up some of that now. At this point, I have found Pi-Fi, and that open source project basically gives me all the functionality I need to use my .zim files with Kiwix.   Second choice after that (for anyone else reading this) would be to buy $30 travel router, that has AP mode and attach a small SSD.  Problem solved.If what you guys are working on does something better great.  I’ll watch for it. But it needs to come out before it’s D.O.A. and replaced by a number of other very simple solutions.sj80231

commit 58e31c79 (2024-04-04): Fixed #68: Check part3 filesystem before and after resize2fs to ensure it succeeeds