offspot/base-image #73: Fix random issues around Ethernet networking

benoit74, 2025-05-05:
When building hotspots, on first startups I immediately plug an Ethernet cable. 

For 1 out of the 3 devices I built today, the hw clock failed to update on first real (second actually) startup (data at  http://hwclock.kiwix.hotspot/ page was wrong).

Then on second start, the captive portal suddenly reappeared only few secs after I closed the first one (I just had time to refresh http://hwclock.kiwix.hotspot/ page which showed that on-board clock was still wrong) and once I confirmed this second captive portal the hwclock page showed that RTC clock has been updated.

I already experienced the same kind of issue (not that precise) on another device.

kelson42, 2025-05-05:
@rgaudin @anupulu Please prioritise this high. The appliance provisioning is already too long, we have to achieve to be quicker/efficientier/robuster.

benoit74, 2025-05-05:
This is does not have very significant impact on provisioning duration tbh. But it definitely make the provisioning process less robust / more complex to explain to a third party. And it probably impact our users as well when they use the hotspot with Internet available on the Ethernet cable).

rgaudin, 2025-05-05:
> When building hotspots, on first startups I immediately plug an Ethernet cable.

Not relevant to this ticket but out of curiosity, why immediately plugging the cable and not plugging before turning on?

> For 1 out of the 3 devices I built today, the hw clock failed to update on first real (second actually) startup (data at http://hwclock.kiwix.hotspot/ page was wrong).

What do you mean by “hw clock failed to update”? Did system clock from network not work or HW clock from system clock?

> Then on second start, the captive portal suddenly reappeared only few secs after I closed the first one (I just had time to refresh http://hwclock.kiwix.hotspot/ page which showed that on-board [...]

That looks like an expected consequence of a late sync.

Assuming it was an NTP sync delay, I suggest  we add a button to HWclock page to trigger a sync manually. It's already possible to set it manually from that page but just requesting update from NTP is less prone to error of course

It's not clear here and I don't think I ever gave clear instructions but H1 assembly should include manually recording system date (has to be correct) into HW if it's not OK. Procedure only mentions to check. It will happen eventually but we cannot rely on an async process that we have no control on at this stage.

@benoit74 do you feel that a trigger button would solve your issue? If so please rename the ticket.

benoit74, 2025-05-05:
> Not relevant to this ticket but out of curiosity, why immediately plugging the cable and not plugging before turning on?

I don't get what the difference is. Once device is assembled, I plug the ethernet cable and then plug power.

> What do you mean by “hw clock failed to update”? Did system clock from network not work or HW clock from system clock?

hw clock was still set to the rubish very old value (don't recall exact value) ; system clock was still set to the image build time

> Assuming it was an NTP sync delay, I suggest we add a button to HWclock page to trigger a sync manually.

I can't be 100% sure, I did not performed sufficient checks, but the fact that captive portal seemed to open twice forces me to think this will probably solve the issue at all. Such a button can't do any bad, it probably even has valid use cases, but I'm pretty sure that NTP sync will still fail when I will click the button. My impression is that network stack is wrong at that moment and no packet goes to/from ethernet cable. Asking for an NTP sync will fail. I should try to open an Internet link to check connectivity, I will do it next time I encounter this bug (hopping I will be fast enough).

> It's not clear here and I don't think I ever gave clear instructions but H1 assembly should include manually recording system date (has to be correct) into HW if it's not OK. Procedure only mentions [...]

The sync of system clock to hw clock always happened in H1 assemblies I've made when system clock was updated with NTP. I feel like forcing a manual sync from system to hardware is only going to cause issues, because the risk that system is off by few hours / days is significant especially once we will start to build custom images. The fact that hw clock is completely wrong is very convenient to quickly spot that there is an issue, and I'm afraid that saying we can force manual sync is going to lead to mistakes.

rgaudin, 2025-05-05:
> My impression is that network stack is wrong at that moment and no packet goes to/from ethernet cable. Asking for an NTP sync will fail. I should try to open an Internet link to check connectivity, I will do it next time I encounter this bug (hopping I will be fast enough).

Ethernet configuration is not changed during your whole procedure. It's set for DHCP and that is set well before docker (and this the portal and all) is started.

It's possible there was some network issue at that moment between the system and the NTP server but from experience, that sync is sometimes weird and we have no control over it. It updates when it wants ; when it thinks it's online. If this is off for some reason, you spend a little time out of sync. That's why I suggested forcing a _resync_

> The fact that hw clock is completely wrong is very convenient to quickly spot that there is an issue

That's quite important to know whether you need to act or not but if it's not, I would say it's better have the ability to force it rather than waiting for some event that might never come (should network be down or whatever).