offspot/base-image #74: Enable RTC battery charging?

rgaudin, 2025-05-22:
Pi5 has a bundled RTC clock. Other Pi can have optional RTC module.
Those module or the Pi5 clock can have an optional battery so the RTC is of any use.

Batteries can be of many kinds, for our H1, we use the [Pi5 Official](https://www.raspberrypi.com/products/rtc-battery/), **rechargeable** LMO one.

The **Pi5** includes a constant current battery charger (1,3mA – 4.4mA) that is [disabled by default](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#enable-battery-charging).

It's not indicated but I imagine it's not ON by default because people could be using non-rechargeable batteries there. Or rechargeable which would not be happy (that can mean **explosion**) with a constant charging? What value to charge it at?

- Should we enable the charger?
- At which current? 3mA seems appropriate for that battery we are using on H1.
- Should we enable it conditionally? Which condition?

Note that enabling for users with no HW clock or no battery has no effect. Users with HAT HW Clock module, be it rasClock or general UPS modules like piSugar or PiJuice are not concerned as those are not plugged to that connector.

The only concern is regarding **Pi5** users using other batteries than the Official one (which we know is rechargeable lithium manganese).

kelson42, 2025-05-22:
Enabling the charging when we deliver kiwix-hotspot with cheargeable battery seems necessary. For all other scenarios I would not, but then it would be maybe interesting to somehow document the situation (but not sure how exactly).

rgaudin, 2025-05-23:
OK, I read this as conditional, with condition being a Kiwix Hotspot H1. We have no way to detect this so we'll have to discuss how to trigger it.

kelson42, 2025-05-27:
> OK, I read this as conditional, with condition being a Kiwix Hotspot H1. We have no way to detect this so we'll have to discuss how to trigger it.

Yes, and it seems to me we better need a solution outside the image. I can imagine two approaches:
* Somehow be able to keep the information in the BIOS
* A post image writing operation (so mounting and modifying the fs)

closed 2025-08-22 by rgaudin